Longan-Labs/Arduino_CAN_BUS_MCP2515 #12: Error Sending Messages...

yadavravi079, 2016-11-16:
Hi, I am having a trouble ... as if am using CAN shield in Loopback mode then I am receiving the msgs transmitted which confirms that all my connections are fine :

Enter Configuration mode successful!
Setting Baudrate Successful!
MCP2515 Initialized Successfully!
MCP2515 Library Loopback Example...
Message Sent Successfully!
Extended ID: 0x00000100  DLC: 8  Data: 0xAA 0x55 0x01 0x10 0xFF 0x12 0x34 0x56
Message Sent Successfully!
Extended ID: 0x00000100  DLC: 8  Data: 0xAA 0x55 0x01 0x10 0xFF 0x12 0x34 0x56
Message Sent Successfully!
Extended ID: 0x00000100  DLC: 8  Data: 0xAA 0x55 0x01 0x10 0xFF 0x12 0x34 0x56
Message Sent Successfully!
Extended ID: 0x00000100  DLC: 8  Data: 0xAA 0x55 0x01 0x10 0xFF 0x12 0x34 0x56
Message Sent Successfully!
Extended ID: 0x00000100  DLC: 8  Data: 0xAA 0x55 0x01 0x10 0xFF 0x12 0x34 0x56
Message Sent Successfully!
....

But I am getting :
 
Enter Configuration mode successful!
Setting Baudrate Successful!
MCP2515 Initialized Successfully!
Error Sending Message...
Error Sending Message...
Error Sending Message...
Error Sending Message...
Error Sending Message...
Error Sending Message...

 if using in 'Normal mode' when i tried to send data 'using CAN_send' sketch.... i am using CoryJFowler's library for can shield.....  *Can anyone please help what wrong with it*????????

coryjfowler, 2016-11-16:
What do you have the shield attached to?  If you do not have another CAN device attached, transmission will fail because no other device is sending an acknowledge.  Otherwise, check your wiring and bitrate.

yadavravi079, 2016-11-16:
The shield is attached to PCAN-USB (via D-sub 9 Connector) and using PCAN-view to monitor the transmit and receive messages.

> check your wiring and bitrate.

Now I am getting the 'message sending successful' but it is not showing the data sent. And also I am not able to receive it  neither be able trace in receiving side of PCAN-view. 
Its only transmitting but not recieving.
 
I tried changing the bitrates ex: 125, 250, 500, ... but still its the same.

coryjfowler, 2016-11-16:
The shield wont show the data it has sent, but it should show up in PCAN-View.  I have tested my fork library extensively with a PCAN-USB dongle for the record.

yadavravi079, 2016-11-17:
ya mistakenly I wrote that it is not showing the data sent, it should show on PCAN-USB dongle. I figured out what wrong I was doing. It was basically related with the transceiver would not be able to get enough voltage supply. I am done with everything is working perfect!!!!!.

Thanks anyways!!!

closed 2016-11-17 by yadavravi079