Longan-Labs/Arduino_CAN_BUS_MCP2515 #13: MCP_CAN on ESP8266 NodeMCU - SPI not correct?

mega-hz, 2017-01-14:
Hello,

on an Arduino Uno i succesfully realized a MCP2515 Shield running perfectly.
Now i am using a ESP8266 with the following SPI Devices:
1x IL9341 TFT, 1x 74HC595 Expander (4x74HC595) and an MCP2515 CAN BUS Shield.

TFT is working fine, also the 74HC595's , the MCP2515 works fine also,
BUT: There is a problem using the 74HC595 AND the MCP2515 together!
If i don't use the MCP, the 74HC595 are working fine, if the MCP2515 is also running, there are always flickering (on the output-LEDs) of the 74HC595!
I can fix it with using spi.begin and spi.end in the 74HC595 Part, but then the MCP is not working anymore!
Is there something missing initializing the SPI on the MCP_CAN Library?
It would be the right way for a spi-device to begin and end, but the mcp_can seems not to do so!

Please can you help me?

regards,
Wolfram.

coryjfowler, 2017-01-16:
If either of your SPI peripherals are using an SPI mode other than 00, you will need to switch it back before doing any CAN instructions.  The begintransaction() and endtransaction() functions are relatively new to the Arduino IDE code set and not necessary in most cases.  They will be added to the library in the future.

mega-hz, 2017-01-17:
thanx,
i use different io's for the shift-register, no spi.
now it works!