Longan-Labs/Arduino_CAN_BUS_MCP2515 #15: Second Mask and Filters Not Working

knopfcol, 2017-07-10:
I am pulling CAN messages off a vehicle based off their CAN ID. I am able to get the first mask to work which corresponds to the first 2 filters. The issue is with the second mask and second set of filters (3-6). Is there something in the library I need to change to enable both masks?

I have attached code for reference. 

Thank you. 

`#include <mcp_can.h>
#include <SPI.h>

// ************************ Section Below for LCD ********************************
//********************************************************************************
#include <Wire.h>
#include <LiquidCrystal_I2C.h>

#define I2C_ADDR    0x27  // Define I2C Address where the PCF8574A is
                          // Address can be changed by soldering A0, A1, or A2
                          // Default is 0x27
                          
// map the pin configuration of LCD backpack for the LiquidCrystal class
#define BACKLIGHT_PIN 3
#define En_pin  2
#define Rw_pin  1
#define Rs_pin  0
#define D4_pin  4
#define D5_pin  5
#define D6_pin  6
#define D7_pin  7

LiquidCrystal_I2C lcd(I2C_ADDR,
                      En_pin,Rw_pin,Rs_pin,D4_pin,D5_pin,D6_pin,D7_pin,
                      BACKLIGHT_PIN, POSITIVE);

// ************************* End LCD Setup ***************************************** 
//********************************************************************************** 

MCP_CAN CAN0(10);                          // Set CS to pin 10

// *****************  Variable declarations *********************************
double Special1;
double Special2;

float RWHL_Spd; // average of rears engine Veh_V_ActlEng sb 48, len 16, res 0.01 offset = 0 kph 0x202 EngVehicleSpThrottle2

// Trail Control Section

float TrlCtlSetSpd;
int BangBit; // Tom: message 0x76, start bit 32, length 1 forma name BrkCtrFnd_B_Stat
int lsmcstate;
int MSR_Status; // active = 1
int rtt;
int RPMStatus;
int TrailCtlStat = 0;
long unsigned int rxId;
unsigned char len = 0;
unsigned char rxBuf[8];
char msgString[128];                        // Array to store serial string

void setup() 

{
  //******************** Section for LCD Initialization ****************************
 
  lcd.begin(20,4);        // 20 columns by 4 rows on display

  lcd.setBacklight(HIGH); // Turn on backlight, LOW for off

// *********************** Misc Set up *******************************************
    pinMode(8,OUTPUT);//Setup Pin 8 for and LED for BangBit:center found by Brakes
  
// *******************************************************************************
[... 195 lines cut ...]
      {
        lcd.print("ENABLED - NOT ACTIVE");
      }
               
    if (rtt ==3)
    {
        lcd.print("OVERSPEED              ");
    }

  // debug items
  //lcd.setCursor(0,3);
  //lcd.print(Special1);
  //lcd.setCursor(10,3);
  //lcd.print(TrailCtlStat);
  //lcd.setCursor(0,3);
  //lcd.print(TrailCtlStat);
  //lcd.setCursor(10,3);
  //lcd.print(rtt);
  //lcd.setCursor(0,1);
  //lcd.print(lsmcstate);
    
}          
}

`

knopfcol, 2017-07-12:
I made progress by placing the following code within the IF statements that pertain to them.

lcd.setCursor(0,0);
lcd.print("Set Speed");
lcd.setCursor(11,0);
lcd.print(" ");
lcd.setCursor(0,3);
lcd.print(" ");
lcd.setCursor(11,0);
lcd.print(TrlCtlSetSpd,0);
lcd.setCursor(0,3);
lcd.print(RPMStatus);

Along with this I eliminated the printing of blanks to erase previous text. This now applies all the filters to the canbus. 

It is interesting though because I receive some messages much more frequently than others even though they should be sending at the same rate. 

Could I just be overloading the arduino with how many messages it is receiving?

coryjfowler, 2017-07-12:
> I am able to get the first mask to work which corresponds to the first 2 filters. The issue is with the second mask and second set of filters (3-6). Is there something in the library I need to change to enable both masks?

There is no way to disable certain masks or filters.  It is possible your program is doing too much in the receive if statement that messages are overflowing the first receive buffer in the protocol controller which moves that message to the second buffer, overwriting its contents.  If you want to turn this off, you will need to modify the .cpp file of the library to disable the bukt bit, see below.   

            //Change this
            mcp2515_modifyRegister(MCP_RXB0CTRL,
            MCP_RXB_RX_MASK | MCP_RXB_BUKT_MASK,
            MCP_RXB_RX_ANY | MCP_RXB_BUKT_MASK);

            //To this
            mcp2515_modifyRegister(MCP_RXB0CTRL,
            MCP_RXB_RX_MASK | MCP_RXB_BUKT_MASK,
            MCP_RXB_RX_ANY);

> Could I just be overloading the arduino with how many messages it is receiving?

That is definitely possible.  You are also updating the LCD with every received message.  There are places where elseif statements would improve performance.  I'd suggest implementing a modified version of the "blink without delay" sketch included with the Ardunio IDE to update your LCD at a specific rate outside of the receive if statement.

> It is interesting though because I receive some messages much more frequently than others even though they should be sending at the same rate.

I've explored CAN on several vehicles and I've never seen messages come at the same rate.  Some are 10Hz, some at 1Hz and a few at 20Hz or faster.