Longan-Labs/Arduino_CAN_BUS_MCP2515 #22: RobotDyn Shield

SketchThis, 2020-01-09:
Hi there! I'm trying to get this CAN bus shield working:

https://robotdyn.com/can-bus-shield-for-arduino.html

I tried the Arduino library from here, burned it onto an Uno and mounted the shield. In the Arduino serial monitor all I get is this:

[image]

I found a fork of this library:

https://github.com/coryjfowler/MCP_CAN_lib by @coryjfowler 

I setup the library and ran the "loopback" example file and I got this:

[image]

So this seems like I'm heading in the right direction, however when I tried some of the other examples I wasn't able to get any live data from my car. I believe I had the car's baudrate set properly... Not totally sure where I need to start looking here... Any help would be greatly appreciated!

Ash-S, 2023-07-14:
Hi @coryjfowler @SketchThis I am using CAN-BUS Shield on top of Arduino Uno. I am facing the similar errors like above.

Message Sent Successfully!
Message Sent Successfully!
Entering Configuration Mode Failure...
Error Initializing MCP2515...
MCP2515 Library Loopback Example...
Message Sent Successfully!
Message Sent Successfully!

Even I tried using the library which you mentioned "https://github.com/coryjfowler/MCP_CAN_lib by @coryjfowler"
but no luck. please help! Thanks.

coryjfowler, 2023-07-16:
I suspect the chip select pin on the shield is not at the default position for the library, this can be changed in the sketch to accommodate the shield.  I cannot access robotdyn[dot]com at the moment, so I can't see what the shield looks like and suggest the correct pin for the chip select.

Ash-S, 2023-07-17:
Hi @coryjfowler , Thanks for your comment.
I guess, 'robotdyn' stopped its production but some information you can see on here. http://useshop.co.kr/shop/item/5355132/
This shows the CS pin 10 as shown on the backside of the board in this link.
I tested it with CS_Pin(10) but it gives me same result. Could you please share a basic test code to check this, if any?
Thanks.

coryjfowler, 2023-07-25:
The loopback example is the basic test.  At this point assuming we are still using an Uno or similar where the SPI pins appear at both 11-13 and ISCP header, then I would suspect that the shield has a clone or fraudulent part that does not function as we and the library expect it.