Longan-Labs/Arduino_CAN_BUS_MCP2515 #31: error compiling for ESP32 dev module

protttoy, 2023-01-20:
Error compiling for ESP32 board. ESP32 Arduino core 2.0.6. Windows 11 64bit. Arduino IDE 1.8.19

Error massage from Arduino:

> 
> C:\Users\CredoSense Desktop 2\Documents\Arduino\libraries\Aruino_CAN_BUS_MCP2515-master\mcp_can.cpp: In member function 'byte MCP_CAN::setMsg(long unsigned int, byte, byte, byte, byte*)':
> 
> C:\Users\CredoSense Desktop 2\Documents\Arduino\libraries\Aruino_CAN_BUS_MCP2515-master\mcp_can.cpp:751:47: error: no matching function for call to 'min(byte&, int)'
> 
>      dta_len     = min(len, MAX_CHAR_IN_MESSAGE);
> 
>                                                ^
> 
> In file included from c:\users\credosense desktop 2\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\algorithm:62,
> 
>                  from C:\Users\CredoSense Desktop 2\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\cores\esp32/Arduino.h:162,
> 
>                  from C:\Users\CredoSense Desktop 2\Documents\Arduino\libraries\Aruino_CAN_BUS_MCP2515-master\mcp_can_dfs.h:27,
> 
>                  from C:\Users\CredoSense Desktop 2\Documents\Arduino\libraries\Aruino_CAN_BUS_MCP2515-master\mcp_can.h:27,
> 
>                  from C:\Users\CredoSense Desktop 2\Documents\Arduino\libraries\Aruino_CAN_BUS_MCP2515-master\mcp_can.cpp:47:
> 
> c:\users\credosense desktop 2\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits\stl_algo.h:3456:5: note: candidate: 'template<class _Tp, class _Compare> _Tp std::min(std::initializer_list<_Tp>, _Compare)'
> 
>      min(initializer_list<_Tp> __l, _Compare __comp)
> 
>      ^~~
> 
> c:\users\credosense desktop 2\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits\stl_algo.h:3456:5: note:   template argument deduction/substitution failed:
> 
> C:\Users\CredoSense Desktop 2\Documents\Arduino\libraries\Aruino_CAN_BUS_MCP2515-master\mcp_can.cpp:751:47: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned char'
> 
>      dta_len     = min(len, MAX_CHAR_IN_MESSAGE);
> 
>                                                ^
> 
> In file included from c:\users\credosense desktop 2\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\algorithm:62,
> 
>                  from C:\Users\CredoSense Desktop 2\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\cores\esp32/Arduino.h:162,
> 
>                  from C:\Users\CredoSense Desktop 2\Documents\Arduino\libraries\Aruino_CAN_BUS_MCP2515-master\mcp_can_dfs.h:27,
> 
>                  from C:\Users\CredoSense Desktop 2\Documents\Arduino\libraries\Aruino_CAN_BUS_MCP2515-master\mcp_can.h:27,
> 
>                  from C:\Users\CredoSense Desktop 2\Documents\Arduino\libraries\Aruino_CAN_BUS_MCP2515-master\mcp_can.cpp:47:
> 
> c:\users\credosense desktop 2\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits\stl_algo.h:3450:5: note: candidate: 'template<class _Tp> _Tp std::min(std::initializer_list<_Tp>)'
> 
>      min(initializer_list<_Tp> __l)
> 
>      ^~~
> 
> c:\users\credosense desktop 2\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits\stl_algo.h:3450:5: note:   template argument deduction/substitution failed:
> 
> C:\Users\CredoSense Desktop 2\Documents\Arduino\libraries\Aruino_CAN_BUS_MCP2515-master\mcp_can.cpp:751:47: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned char'
> 
>      dta_len     = min(len, MAX_CHAR_IN_MESSAGE);
> 
>                                                ^
> 
> In file included from c:\users\credosense desktop 2\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\algorithm:61,
> 
>                  from C:\Users\CredoSense Desktop 2\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\cores\esp32/Arduino.h:162,
> 
>                  from C:\Users\CredoSense Desktop 2\Documents\Arduino\libraries\Aruino_CAN_BUS_MCP2515-master\mcp_can_dfs.h:27,
> 
>                  from C:\Users\CredoSense Desktop 2\Documents\Arduino\libraries\Aruino_CAN_BUS_MCP2515-master\mcp_can.h:27,
> 
>                  from C:\Users\CredoSense Desktop 2\Documents\Arduino\libraries\Aruino_CAN_BUS_MCP2515-master\mcp_can.cpp:47:
> 
> c:\users\credosense desktop 2\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits\stl_algobase.h:243:5: note: candidate: 'template<class _Tp, class _Compare> const _Tp& std::min(const _Tp&, const _Tp&, _Compare)'
> 
>      min(const _Tp& __a, const _Tp& __b, _Compare __comp)
> 
>      ^~~
> 
> c:\users\credosense desktop 2\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits\stl_algobase.h:243:5: note:   template argument deduction/substitution failed:
> 
> C:\Users\CredoSense Desktop 2\Documents\Arduino\libraries\Aruino_CAN_BUS_MCP2515-master\mcp_can.cpp:751:47: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned char' and 'int')
> 
>      dta_len     = min(len, MAX_CHAR_IN_MESSAGE);
> 
>                                                ^
> 
> In file included from c:\users\credosense desktop 2\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\algorithm:61,
> 
>                  from C:\Users\CredoSense Desktop 2\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\cores\esp32/Arduino.h:162,
> 
>                  from C:\Users\CredoSense Desktop 2\Documents\Arduino\libraries\Aruino_CAN_BUS_MCP2515-master\mcp_can_dfs.h:27,
> 
>                  from C:\Users\CredoSense Desktop 2\Documents\Arduino\libraries\Aruino_CAN_BUS_MCP2515-master\mcp_can.h:27,
> 
>                  from C:\Users\CredoSense Desktop 2\Documents\Arduino\libraries\Aruino_CAN_BUS_MCP2515-master\mcp_can.cpp:47:
> 
> c:\users\credosense desktop 2\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits\stl_algobase.h:195:5: note: candidate: 'template<class _Tp> const _Tp& std::min(const _Tp&, const _Tp&)'
> 
>      min(const _Tp& __a, const _Tp& __b)
> 
>      ^~~
> 
> c:\users\credosense desktop 2\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits\stl_algobase.h:195:5: note:   template argument deduction/substitution failed:
> 
> C:\Users\CredoSense Desktop 2\Documents\Arduino\libraries\Aruino_CAN_BUS_MCP2515-master\mcp_can.cpp:751:47: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned char' and 'int')
> 
>      dta_len     = min(len, MAX_CHAR_IN_MESSAGE);
> 
>                                                ^
> 
> Using library SPI at version 2.0.0 in folder: C:\Users\CredoSense Desktop 2\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\SPI 
> 
> Using library Aruino_CAN_BUS_MCP2515-master at version 1.0.0 in folder: C:\Users\CredoSense Desktop 2\Documents\Arduino\libraries\Aruino_CAN_BUS_MCP2515-master 
> 
> exit status 1
> 
> Error compiling for board ESP32 Dev Module.
> 
>

juliandower, 2023-01-30:
I managed to fix it using a suggestion from ChatGPT:

The error message says that the function "min" cannot be used with a parameter of type 'byte' and a parameter of type 'int'. This function is used in "mcp_can.cpp" line 751. The issue seems to be caused by the fact that the "min" function is being passed a "byte" and an "int". To resolve this issue, you could cast the "int" to a "byte". Try changing this line:

dta_len = min(len, MAX_CHAR_IN_MESSAGE);

to this:

dta_len = min(len, (byte)MAX_CHAR_IN_MESSAGE);

sidney2007, 2023-11-05:
Already fixed in https://github.com/Longan-Labs/Arduino_CAN_BUS_MCP2515/commit/b62f62065506a30f6eb3aa0f2c7b0757f8d57db9

closed 2023-11-05 by sidney2007